kevinpt/symbolator #11: Unable to install the dependencies on windows

sumanth-kalluri, 2020-07-14:
I have spent days trying to figure out how to get this thing working. I have python 2 and python 3 installed on windows.
I've also downloaded the latest GTK libraries. All I get when I run Symbolator from the command line is 
``` no module named: 'cairo' ```
when I do ``` pip install pycairo ``` all I get is ``` cairo.h : No such file or directory found ``` although I can clearly find a cairo.h on using the search in the file manager. I eventually gave up on windows and tried installing it on Ubuntu using WSL and although I was able to take care of the 'cairo' dependency somehow I'm still getting a ``` no module named: 'pango' ```
I'm unable to find a pango in Pypi or anywhere else. 

This is extremely frustrating. Please help me.

nobodywasishere, 2020-07-14:
If you want to try using python3 (and pip3) you can take a look at my fork [here](https://github.com/nobodywasishere/symbolator). I fixed a few errors I was running into while trying to get it working, and moved everything to python3. You'll need to install python3 and pip3, and install these dependencies from pip3 (`hdlparse pycairo pygobject`). From there, you should be able to do `python3 ./symbolator.py` and have it either find a *.vhdl file or it'll say `Please provide a proper input file`. Make sure if you're using python3 though that you're installing everything from pip3 and not pip. If it's still unable to find pango, try installing `libpango-1.0-0` from apt (I couldn't find a pango build for windows, sorry). If you run into any errors, please let me know, as it's possible I missed something. This was tested on Ubuntu 18.04.4.

sumanth-kalluri, 2020-07-15:
Hi @nobodywasishere!

First of all thanks a lot for replying. I was able to get this to work after installing ```gir1.2-pango-1.0``` from apt, in addition to everything you mentioned. This was pointed out in some other git issue. Before that I was getting a ```ValueError: Namespace PangoCairo not available``` message. I'm on ubuntu 20.04 LTS

Thank you!