kevinpt/symbolator #12: Got this error while running symbolator on a verilog file

sumanth-kalluri, 2020-07-15:
Is symbolator properly supported for verilog? I got the following error when I ran it on a .v file. I have tried out multiple output formats etc. but no joy.

``` sumanth@LAPTOP-MFGQBV4B:/mnt/g/symbolator_files/symbolator$ python3
./symbolator.py alu.v
Scanning library: .                                                                                                    
Creating symbol for alu.v "alu"
-> __alu.svg              
Traceback (most recent call last):
File "./symbolator.py", line 596, in <module>                                                                             
   main()                                                                                                                
File "./symbolator.py", line 591, in main                                                                                 
   sym.draw(0,0, nc)                                          
File "./symbolator.py", line 272, in draw                                                                                 
   sym_width = max(s.min_width(c, style.font) for sym in self.symbols for s in sym.sections)
File "./symbolator.py", line 272, in <genexpr>                                                                            
   sym_width = max(s.min_width(c, style.font) for sym in self.symbols for s in sym.sections)
File "./symbolator.py", line 158, in min_width                                                                            
   lmax = max(tw.text_width(c, font_params) for tw in self.left_pins)
File "./symbolator.py", line 158, in <genexpr>                                                                            
   lmax = max(tw.text_width(c, font_params) for tw in self.left_pins)
File "./symbolator.py", line 105, in text_width                                                                           
   x0, y0, x1, y1, baseline = c.surf.text_bbox(self.text, font_params)
File "/mnt/g/symbolator_files/symbolator/nucanvas/svg_backend.py", line 211, in text_bbox                                 
   return CairoSurface.cairo_text_bbox(text, font_params, spacing, self.scale)
File "/mnt/g/symbolator_files/symbolator/nucanvas/cairo_backend.py", line 122, in cairo_text_bbox
   layout = pangocairo.create_layout(ctx)
KeyError: 'could not find foreign type Context'
sumanth@LAPTOP-MFGQBV4B:/mnt/g/symbolator_files/symbolator$ ```

nobodywasishere, 2020-07-15:
I know with VHDL, the component needs to be within a package for it to work, and I'm not sure what the verilog equivalent is. If you're using the fork I made though, I'm still trying to figure out how everything works and could have messed it up. I need to do more tests on verilog code

SiliconKite, 2020-07-22:
how does this verilog file look like? does it use c-style ports? does it use any SV (maybe interfaces?) feature? can you post it here? (at least the module declaration)

sumanth-kalluri, 2020-07-22:
sure. Here is the complete file. It instantiates one module which is a shift register. I don't think it has anything out of the ordinary

```verilog
`timescale 1ns / 1ps
module alu #(parameter bit_width = 8,parameter ip_size = 16, parameter window_size = 3)(
    input clk,
    input rst,
    input clk_en,
    input [bit_width-1:0] data,
    input wire [71:0] weight_array,
    output wire [17:0] sum_out
    );
    
    wire [16:0] sum;
    wire [7:0] weights [0:8];    
    wire [bit_width*(ip_size+2)-1:0] reg1_op,reg2_op;
    wire [bit_width*window_size-1:0] reg3_op;
    reg [15:0] product [0:8];
    reg ce;
    
      generate 
	  genvar unpk_idx;  
	  for (unpk_idx=0; unpk_idx<(9); unpk_idx=unpk_idx+1) 
	  begin:unpack1
	  assign weights[unpk_idx][7:0] = weight_array[((8)*unpk_idx) +: 8]; 
	  end 
	  endgenerate
    
    variable_shift_reg #(bit_width,ip_size+2) s1 (clk,ce,rst,data,reg1_op);
    variable_shift_reg #(bit_width,ip_size+2) s2 (clk,ce,rst,reg1_op[bit_width*(ip_size+2)-1:bit_width*(ip_size+1)],reg2_op);
    variable_shift_reg #(bit_width,window_size) s3 (clk,ce,rst,reg2_op[bit_width*(ip_size+2)-1:bit_width*(ip_size+1)],reg3_op);
    
    always@(posedge clk) begin
    ce <= clk_en;
    end
    
    generate
    genvar i;
	 for(i=0;i<9;i=i+1)
    begin:block2
    always@(posedge clk or posedge rst) begin
            if(rst)
            begin
                product[i] <= 16'b0;
            end
            else
				if(ce)     
             begin
                if(i <= 2)
                begin
                    product[i] <= weights[i] * reg1_op[bit_width*(i+1)-1:bit_width*i];
                end 
                else if(i > 2 && i <= 5)
                begin
                    product[i] <= weights[i] * reg2_op[bit_width*(i-3+1)-1:bit_width*(i-3)];
                end
                else if(i > 5)
                begin
                    product[i] <= weights[i] * reg3_op[bit_width*(i-6+1)-1:bit_width*(i-6)];
                end
             end
           end
    end
    endgenerate
    
    assign sum_out = product[0]+product[1]+product[2]+product[3]+product[4]+product[5]+product[6]+product[7]+product[8];
endmodule
```

nobodywasishere, 2020-07-22:
What version of symbolator are you using? It worked for me using my (hopefully temporary) fork and the version currently available in pip. I am using python3 which may be the difference though.

[image]

sumanth-kalluri, 2020-07-22:
Just updated my files from your fork and tried again. Still no luck. The symbolator I downloaded from pip3 has a version ```1.0.2```

sumanth-kalluri, 2020-07-22:
@nobodywasishere If you don't mind could you please tell me the versions of hdlparse, pycairo and pygobject that you have?

nobodywasishere, 2020-07-22:
```
┌─[michael-08:49:47]───[~/dev/symbolator]───[16 files, 76K]
└───5▶ python --version
Python 3.8.3
┌─[michael-08:50:39]───[~/dev/symbolator]───[16 files, 76K]
└───6▶ pip --version
pip 20.1.1 from /home/michael/.local/lib/python3.8/site-packages/pip (python 3.8)
┌─[michael-08:50:44]───[~/dev/symbolator]───[16 files, 76K]
└───7▶ pip show hdlparse
Name: hdlparse
Version: 1.0.5
Summary: HDL parser
Home-page: http://kevinpt.github.io/hdlparse
Author: Kevin Thibedeau
Author-email: kevin.thibedeau@gmail.com
License: MIT
Location: /usr/lib/python3.8/site-packages/hdlparse-1.0.5-py3.8.egg
Requires: 
Required-by: symbolator
┌─[michael-08:51:04]───[~/dev/symbolator]───[16 files, 76K]
└───8▶ pip show pycairo
Name: pycairo
Version: 1.19.1
Summary: Python interface for cairo
Home-page: https://pycairo.readthedocs.io
Author: Christoph Reiter
Author-email: reiter.christoph@gmail.com
License: LGPL-2.1-only OR MPL-1.1
Location: /usr/lib/python3.8/site-packages
Requires: 
Required-by: 
┌─[michael-08:51:12]───[~/dev/symbolator]───[16 files, 76K]
└───9▶ pip show pygobject
Name: PyGObject
Version: 3.36.1
Summary: Python bindings for GObject Introspection
Home-page: https://pygobject.readthedocs.io
Author: James Henstridge
Author-email: james@daa.com.au
License: GNU LGPL
Location: /usr/lib/python3.8/site-packages
Requires: 
Required-by: udiskie, lutris
┌─[michael-08:51:22]───[~/dev/symbolator]───[16 files, 76K]
└───10▶ screenfetch

 ██████████████████  ████████     michael@michael-t450
 ██████████████████  ████████     OS: Manjaro 20.0.3 Lysia
 ██████████████████  ████████     Kernel: x86_64 Linux 5.6.16-1-MANJARO
 ██████████████████  ████████     Uptime: 1d 14h 48m
 ████████            ████████     Packages: 1543
 ████████  ████████  ████████     Shell: bash 5.0.18
 ████████  ████████  ████████     Resolution: 1920x1080
 ████████  ████████  ████████     DE: Xfce4
 ████████  ████████  ████████     WM: Xfwm4
 ████████  ████████  ████████     WM Theme: Adapta-Maia
 ████████  ████████  ████████     GTK Theme: Matcha-dark-sea [GTK2]
 ████████  ████████  ████████     Icon Theme: Papirus-Maia
 ████████  ████████  ████████     Font: Noto Sans 10
 ████████  ████████  ████████     Disk: 220G / 247G (94%)
                                  CPU: Intel Core i5-5300U @ 4x 2.9GHz [50.0°C]
                                  GPU: Mesa Intel(R) HD Graphics 5500 (BDW GT2)
                                  RAM: 3450MiB / 7841MiB
┌─[michael-08:55:00]───[~/dev/symbolator]───[11 files, 76K]
└───45▶ ./symbolator.py --version
Symbolator 1.0.2a
┌─[michael-08:55:08]───[~/dev/symbolator]───[11 files, 76K]
└───46▶ ./symbolator.py doc/images/demo_device.vhdl 
Scanning library: .
Creating symbol for doc/images/demo_device.vhdl "demo_device"
	-> demo_device-demo_device.svg
┌─[michael-08:55:41]───[~/dev/symbolator]───[12 files, 80K]
└───47▶ pip show symbolator
Name: symbolator
Version: 1.0.2
Summary: HDL symbol generator
Home-page: http://kevinpt.github.io/symbolator
Author: Kevin Thibedeau
Author-email: kevin.thibedeau@gmail.com
License: MIT
Location: /usr/lib/python3.8/site-packages
Requires: hdlparse
Required-by: 
┌─[michael-08:55:53]───[~/dev/symbolator]───[12 files, 80K]
└───48▶ symbolator --version
/usr/lib/python3.8/site-packages/nucanvas/cairo_backend.py:17: PyGIWarning: Pango was imported without specifying a version first. Use gi.require_version('Pango', '1.0') before import to ensure that the right version gets loaded.
  from gi.repository import Pango as pango
/usr/lib/python3.8/site-packages/nucanvas/cairo_backend.py:18: PyGIWarning: PangoCairo was imported without specifying a version first. Use gi.require_version('PangoCairo', '1.0') before import to ensure that the right version gets loaded.
  from gi.repository import PangoCairo as pangocairo
Symbolator 1.0.2
┌─[michael-08:55:59]───[~/dev/symbolator]───[12 files, 80K]
└───49▶ symbolator doc/images/demo_device.vhdl 
/usr/lib/python3.8/site-packages/nucanvas/cairo_backend.py:17: PyGIWarning: Pango was imported without specifying a version first. Use gi.require_version('Pango', '1.0') before import to ensure that the right version gets loaded.
  from gi.repository import Pango as pango
/usr/lib/python3.8/site-packages/nucanvas/cairo_backend.py:18: PyGIWarning: PangoCairo was imported without specifying a version first. Use gi.require_version('PangoCairo', '1.0') before import to ensure that the right version gets loaded.
  from gi.repository import PangoCairo as pangocairo
Scanning library: .
Creating symbol for doc/images/demo_device.vhdl "demo_device"
	-> demo_device-demo_device.svg
┌─[michael-08:56:21]───[~/dev/symbolator]───[12 files, 80K]
└───50▶ 
```

I want to test this in a fresh Ubuntu VM to see if I can reproduce the error. What version of Ubuntu are you using?

sumanth-kalluri, 2020-07-22:
@nobodywasishere I updated my packages (pygobject,hdlparse) to the versions in your system and it is now working. 
Thanks a ton for all the help!

mithro, 2020-07-22:
FYI - We have started maintaining a symbolator fork in the SymbiFlow organization (https://github.com/SymbiFlow/symbolator).

We are reviewing and accepting pull requests as the tool is being heavily used in SymbiFlow and [Google's skywater-pdk documentation](https://github.com/google/skywater-pdk). 

At some point we might rename it to something like symbiflow-symbolator and publish it on PyPi and similar.

nobodywasishere, 2020-07-22:
My fork is a move to only Python 3, fixes for some bugs I ran into, and a merge of the changes made by various forks of the original repo. It was more of a short-term test than a long-term maintain. If we can get this updated and rolling again, I'd be more than happy to help. 

I would also fork hdlparse if possible so we can get that rolling too. Stuff like parsing VHDL entities instead of just packages requires changing hdlparse, and there's already code for both in adding this feature waiting to be merged.

mithro, 2020-07-22:
We are mainly using symbolator with verilog and don't have a huge interest in VHDL (but perfectly happy for other people to work on that).

sumanth-kalluri, 2020-07-23:
@mithro what exactly are you planning with Symbolator in the Symbiflow? I think Its high time someone created an open-source HDL to block diagram converter that enables easier design exploration by allowing us to zoom into the sub-modules. Something like the Schematic Viewer in Vivado and ISE. 

If something like that is on the charts. I would love to contribute to the project.

mithro, 2020-07-23:
@sumanth-kalluri - Don't have any plans to dramatically improve things in symbolator, we just need basic block diagrams for things like the skywater-pdk. 

There are other projects working on the schematic viewer idea -- see https://github.com/SymbiFlow/ideas/issues/41 -- I actually think there is a GSoC project related to that...

sumanth-kalluri, 2020-07-23:
@mithro I wish I was still a student. Anyways great to see such interest being taken on this idea.

nobodywasishere, 2020-07-23:
@mithro What are your plans in terms of Python 2 support since Python 2.7 is now EOL? I can do a PR for my changes to Python 3 only on your fork if you want.

mithro, 2020-07-23:
I think supporting Python 3 is important.

mithro, 2020-07-23:
@sumanth-kalluri -- There was also a discussion about this topic at https://groups.google.com/forum/#!msg/eda-playground/Mh9bOhC7FNQ/w7GKEEqlAAAJ and potential of using Surelog to do this -> https://github.com/alainmarcel/Surelog

nobodywasishere, 2020-07-24:
I went through and redid my forks of symbolator and hdlparse in a cleaner fashion. Each has an entity branch specifically for entity support and I updated the version numbers for those branches as to prevent compatibility errors when installing (you'd need to install both entity branches manually for VHDL entities to work). I pulled in all the various changes people have made on their own forks as well. I'll probably maintain these just for personal use. 

I also made a PR to SymbiFlow/symbolator with the changes I specifically made to drop Python 2.7 and fix a PyGIWarning. I hope that works for what you're looking for.

SiliconKite, 2020-07-24:
Thank you guys and thanks @nobodywasishere , this is excellent work

tvannoy, 2020-07-31:
@nobodywasishere Your entity branches worked great for what I am doing. 

I'll try to keep up to date with the development happening here and contribute to the Symbiflow fork.