kevinpt/symbolator #14: pip install doesn't seem work?

SiliconKite, 2020-07-22:
Hi,
eager to try this tool but I can't get it to install. First of all it doesn't automatically install pycairo and pango dependencies? (hdlparse is installed fine though)

Then, if I install them, install symbolator and run the tool, I get the following error :

```
File "/home/${USER}/.local/lib/python3.7/site-packages/nucanvas/cairo_backend.py", line 13, in <module>
    import pango
ModuleNotFoundError: No module named 'pango'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./symbolator", line 11, in <module>
    load_entry_point('symbolator==1.0.2', 'console_scripts', 'symbolator')()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/${USER}/.local/lib/python3.7/site-packages/symbolator.py", line 10, in <module>
    from nucanvas.cairo_backend import CairoSurface
  File "/home/${USER}/.local/lib/python3.7/site-packages/nucanvas/cairo_backend.py", line 17, in <module>
    from gi.repository import Pango as pango
ModuleNotFoundError: No module named 'gi'
```
Further research brings me to the issue that I am missing "gi" package. When installing this package, `pygobject` I run into some other issues ...

Note that I am installing everything on python3 through `pip install --user`, not sure this has something to do with the issue, although I never had an issue so far ...

nobodywasishere, 2020-07-22:
I ran into a lot of issues you're experiencing as well. Try out my (hopefully temporary) fork [here](https://github.com/nobodywasishere/symbolator) and let me know if that works or not.

SiliconKite, 2020-07-23:
Thanks , its working great on Fedora 31 and python3 !