kevinpt/symbolator #18: HDL parse is not good enough

joezhouchenye, 2022-10-14:
I have encountered two situations that symbolator generates an unexpected symbol.

- Instantiation of a module whose name contains input, such as input_buffer. Symbolator will render this as a port "_buffer".
- Comments in a Verilog module's parameter declaration part “#()”. Symbolator will also render this as an unexpected parameter port.

joezhouchenye, 2022-10-14:
As for the first situation, Symbolator is only used to get the port and parameter information. It doesn't need to read the entire file.
Find the site-packages location of python and modify the verilog_parser.py file.

[image]

Change 'endmodule' to ';'

joezhouchenye, 2022-10-14:
As for the third situation, parameters should not be treated as special pins.

I modified the symbolator.py file.

- Modify the make_section function and add a argument pattern with the default value True.

```python
def make_section(sname, sect_pins, fill, extractor, no_type=False, pattern=True):
  '''Create a section from a pin list'''
  sect = PinSection(sname, fill=fill)
  side = 'l'

  for p in sect_pins:
    pname = p.name
    pdir = p.mode
    data_type = p.data_type if no_type == False else None
    bus = extractor.is_array(p.data_type)

    pdir = pdir.lower()

    # Convert Verilog modes
    if pdir == 'input':
      pdir = 'in'
    if pdir == 'output':
      pdir = 'out'

    # Determine which side the pin is on
    if pdir in ('in'):
      side = 'l'
    elif pdir in ('out', 'inout'):
      side = 'r'

    pin = Pin(pname, side=side, data_type=data_type)
    if pdir == 'inout':
      pin.bidir = True

    if pattern:
	    # Check for pin name patterns
	    pin_patterns = {
	      'clock': re.compile(r'(^cl(oc)?k)|(cl(oc)?k$)', re.IGNORECASE),
	      'bubble': re.compile(r'_[nb]$', re.IGNORECASE),
	      'bus': re.compile(r'(\[.*\]$)', re.IGNORECASE)
	    }

	    if pdir == 'in' and pin_patterns['clock'].search(pname):
	      pin.clocked = True

	    if pin_patterns['bubble'].search(pname):
	      pin.bubble = True

	    if bus or pin_patterns['bus'].search(pname):
	      pin.bus = True

    sect.add_pin(pin)

  return sect
```

- In the make_symbol function, change the following.

[image]

Paebbels, 2023-02-05:
See https://github.com/hdl/symbolator as a fork.