kevinpt/symbolator #19: Installation error from pip3 on ubuntu

bat52, 2023-01-13:
I am getting the following error when trying to install on ubuntu 22.04.
Install using pip (python2.7) works fine instead.

```
pip3 install --upgrade symbolator
Defaulting to user installation because normal site-packages is not writeable
Collecting symbolator
  Using cached symbolator-1.0.2.tar.gz (30 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in symbolator setup command: use_2to3 is invalid.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

Paebbels, 2023-02-05:
See https://github.com/hdl/symbolator as a fork.

bat52, 2024-09-26:
> Maybe run pip3 install setuptools==58 in your virtual environment could help.
Thank you ! I confirm this works:
`pip3 install setuptools==58 symbolator`
Shouldn't this be fixed with an update of setup.py ?